Wallacoloo/printipi #65: [Raspberry Pi] Consider using the GPU for precise GPIO timings, rather than DMA

Wallacoloo, 2015-01-03:
Gert showed that it was possible to dump the value of over 50 million 16-bit values per second into the GPIOs using the Pi's GPU. His application was to have each 16 bits present one pixel (RGB565) and use a resistor DAC to send it to a VGA LCD, along with clock signals generated by the GPU, all with accurate enough timings for video (in order to correctly place each pixel, that requires timing accurate to about 20 ns).

Unfortunately, his code, at the time of writing, is only supplied as a binary blob and the documentation of the GPU features he used is nonexistent.

But there are other efforts to demystify the Pi's GPU. There is, for example, [this](http://petewarden.com/2014/08/07/how-to-optimize-raspberry-pi-code-using-its-gpu/) tutorial for writing GPU code on the Pi. It **may** be possible to commandeer one GPU core permanently and use it to precisely drive the GPIO pins (Not sure how much control we have over scheduling).

Wallacoloo, 2015-01-03:
The Thread Control section of [this](http://www.broadcom.com/docs/support/videocore/VideoCoreIV-AG100-R.pdf) document seems to indicate that once a program is allocated space onto one of the GPU cores, it has full control of its scheduling - it can only lose control via an explicit `Thread Switch` signal.

I don't know how memory writes are coordinated. But I imagine there's a way to write to arbitrary memory locations (i.e. the GPIO bank).

Wallacoloo, 2015-01-03:
Started [this](http://www.raspberrypi.org/forums/viewtopic.php?f=37&t=95572) thread on the Raspberry Pi forums. jamesh suggests using the scalar cores.

niksoley, 2016-10-13:
It sounds good, how is it evolving? Is there still the issue of losing steps at high speeds?

Wallacoloo, 2016-10-13:
@niksoley This project's been dead for a while now. I never even started work on the GPU stuff. Feel free to fork & develop it if you like though.