Wallacoloo/printipi #95: Cannot run printipi in Raspberry pi 3

Mokutsuno, 2017-04-03:
hi.
I tried to run printipi but logs below appeared and ended it.
---------------------------------------------------------------------------------
2017/04/03 12:26:40 socat[7107] N PTY is /dev/pts/6
2017/04/03 12:26:40 socat[7107] N PTY is /dev/pts/7
2017/04/03 12:26:40 socat[7107] N starting data transfer loop with FDs [6,6] and [8,8]
Launching Printipi firmware configured to take commands from /dev/ttyUSBprintipi
NOTE: This will appear to hang the console, but that is not an error
Killing the console (Ctrl+c or Ctrl+z) will also kill Printipi & leave /dev/ttyUSBprintipi unresponsive
2017/04/03 12:26:40 socat[7107] N socat_signal(): handling signal 1
2017/04/03 12:26:40 socat[7107] N exiting on signal 1
2017/04/03 12:26:40 socat[7107] N socat_signal(): finishing signal 1
2017/04/03 12:26:40 socat[7107] N exit(129)
---------------------------------------------------------------------------------

I tryed also executing `sudo sh launch-firmware.sh` and printipi worked normally. 
But now, I cannot connect to Octoprint.

Octoprint logs -------------------------------------------------------------------
Connecting to: /dev/ttyUSB0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x6fb3f530, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection
Changing monitoring state from 'Opening serial port' to 'Detecting baudrate'
Trying baudrate: 250000
...
Changing monitoring state from 'Detecting baudrate' to 'Offline'
Changing monitoring state from 'Offline' to 'Error: No more baudrates to test, and no suitable baudrate found.'
Connection closed, closing down monitor
---------------------------------------------------------------------------------

How can I resolve this?

Mokutsuno, 2017-04-05:
I had compiled it again that I accorded to the procedure.
Then My issue is move to [https://github.com/Wallacoloo/printipi/issues/91](url)

Wallacoloo, 2017-04-05:
@Mokutsuno Sorry it's not working on the Pi 3. Without detailed information to work with, I can't recommend any workaround/fixes. I'm not actively developing Printipi, so the only way this will get solved is if somebody else submits a pull request for Pi 2/Pi 3 support.

This bug report doesn't really provide any additional information than #91, and it mostly overlaps it, so I'm going to close this and #91 will serve for tracking Pi 2/Pi 3/ support.

closed 2017-04-05 by Wallacoloo