Wallacoloo/printipi #96: I succeeded to be run printipi on rpi3,But I have got other problem.

Mokutsuno, 2017-05-01:
I succeeded being run this on rpi3 .
I used pigpio.
So,I made new own  primitiveiopin.h , mitpi.h and mitpi.cpp ,which used pigpio.

And I got a problem.
printpi  toggles PWM ,when I send command of "G1 ....".
It moves as intended.
However, To toggle PWM isn't end,that is,printipi doesn't move on next command.
Also "command: G1... "and"response:ok" are not outputted.

[image]

I think that printpi moves on next command ,when "exitEventLoop()" is called.
So, something problem have occured, then "exitEventLoop()" haven't called.

```
//////////////////////////////////////////////////////////////////////
#ifndef PLATFORMS_RPI_PRIMITIVEIOPIN_H
#define PLATFORMS_RPI_PRIMITIVEIOPIN_H

#include <cassert>

#include "mitpi.h" //for GpioPin
//#include "hardwarescheduler.h" //for ability to schedule PWM
#include "platforms/auto/chronoclock.h" //for EventClockT.
//for MAX_RPI_PIN_ID
#include "compileflags.h"
#include <pigpiod_if.h>
namespace plat {
	namespace rpi2 {

		class PrimitiveIoPin {
			//logical index of this pin (0-53, I believe)
			mitpi::GpioPin pinIdx;
		public:
			//@return a pin with an invalid output
			//
			//Note that it is undefined to call any function on a null pin except for <isNull> and <id>.
			inline static PrimitiveIoPin null() {
				return PrimitiveIoPin(mitpi::NULL_GPIO_PIN);
			}
			//@return true if the pin is equal to the one constructed by the <null> function.
			inline bool isNull() const {
				return pinIdx == mitpi::NULL_GPIO_PIN;
			}
			//@pinIdx *logical* index of the pin
			//@pullUpDown direct this pin to either pull up to 3.3v, down to gnd, or no pull at all.
			//Note that this pull direction will be applied **even when operating as an output pin**.
			// The rpi's pull resistors preserve state across reboots.
			// This makes it so that when the Pi resets, its pins will still be in a defined state (no runaway heater while the Pi is still booting)
			// Of course, it's still a VERY GOOD IDEA to use hardware pull resistors as well.
			//**Also note:** @pullUpDown acts the same regardless of the <IoPin>'s read/write inversions.
        template <typename ...T> PrimitiveIoPin(mitpi::GpioPin pinIdx, mitpi::GpioPull pullUpDown = mitpi::GPIOPULL_NONE) {
				assert((pinIdx == mitpi::NULL_GPIO_PIN || pinIdx <= MAX_RPI_PIN_ID) && "Make sure to appropriately set MAX_RPI_PIN_ID (see compileflags.h) or else some pins might not behave correctly");
				if (pinIdx != mitpi::NULL_GPIO_PIN) {
					mitpi::init();
					//we CANNOT set the pin pull of a null gpio pin
					mitpi::setPinPull(pinIdx, pullUpDown);
				}
        }
			//@return the logical index of the pin (for use with other platform-specific functions)
			inline mitpi::GpioPin id() const {
				return pinIdx;
			}
			//configure the pin as an output, and set its output state
			inline void makeDigitalOutput(IoLevel lev) {
				mitpi::makeOutput(pinIdx);
				digitalWrite(lev);
			}
			//configure the pin as a PWM output & set its duty cycle and period (if applicable)
			inline void makePwmOutput(float duty, EventClockT::duration desiredPeriod) {
				mitpi::makeOutput(pinIdx);
				//mitpi::pwmWrite(pinIdx, duty, desiredPeriod);
				pwmWrite(duty, desiredPeriod);
			}
			//configure the pin to be an input
			inline void makeDigitalInput() {
				mitpi::makeInput(pinIdx);
			}
			//read the pin's input value (must call makeDigitalInput beforehand)
			inline IoLevel digitalRead() const {
				return mitpi::readPinState(pinIdx);
			}
			//write a digital value to the pin. IoHigh = 3.3v, IoLow = 0v. Must call makeDigitalOutput beforehand
			inline void digitalWrite(IoLevel lev) {
				mitpi::setPinState(pinIdx, lev);
			}
			//set pwm duty cycle & period (if applicable). Must call makePwmOutput beforehand.
			inline void pwmWrite(float duty, EventClockT::duration desiredPeriod) {
				float minPeriod = std::chrono::duration_cast<std::chrono::duration<float>> (desiredPeriod).count();
				int intMinPeriod = (int)minPeriod;
				uint32_t intDuty = 1000000*duty;
				hardware_PWM(this->id(), intMinPeriod, intDuty);
					//mitpi::pwmWrite(pin, duty, desiredPeriod);
				//HardwareScheduler().queuePwm(*this, duty, desiredPeriod);
			}
		};
	}
}
#endif
///////////////////////////////////////////////////////////////
```
About pwmWrite(), I haven't understood very much yet.

You said you not actively developing Printipi ,so please reply to me, if it's OK with you..

Wallacoloo, 2017-05-03:
You haven't given me much information to work on, here. I would suggest you run Printipi under a debugger (like `gdb`). You could interrupt it when it hangs (with Ctrl+C) and then get a stack trace by typing `bt`. This might reveal if it's stuck in a loop / blocking call, etc.

One thing I would also do is check the return code of `hardware_PWM`. I'm not sure which parts of the code you've edited, but you should make sure that the original [hardwarescheduler.cpp](https://github.com/Wallacoloo/printipi/blob/master/src/platforms/rpi/hardwarescheduler.cpp) isn't being compiled - it also uses DMA and might conflict with the DMA channel that pigpio's `hardware_PWM` function is using.

Wallacoloo, 2017-05-03:
I just dug up the documentation for [hardware_PWM](http://abyz.co.uk/rpi/pigpio/pdif2.html#hardware_PWM). It looks to me like this is calling into a separate daemon to manage the gpios. As such, if you haven't started the pigpio daemon before running printipi, it's quite possible that their code is written in a way where it stalls until the daemon is started.

Mokutsuno, 2017-05-03:
Thank you for your reply.

printipi doesn't hang. printipi executes first command (G1 Y0.1 F6000 e.g.),but doesn't execute next command. It keeps outputting "[inf0] RCThermistor2Pin read timeout" during running it.

The original hardwarescheduler.cpp isn't being compiled.
The pigpio daemon has been start since boot the raspberry pi 3.

I used gdb. As a result,`hardware_PWM()` did not return 0. It returned -95.
I noticed that  `hardware_PWM(`)(`pwmWrite()`) was called twice only.
First,it is called about V1_GPIO_P1_08 (probably FAN).
Second,called about V1_GPIO_P1_10 (probably HOTEND). Both return -95.
`hardware_PWM(`)(`pwmWrite()`)  isn't called, when I input commands.
I can't understand why GPIO toggle (But as above,  it doesn't work as intended). There are no functions to make PWM other than `hardware_PWM(`)(`pwmWrite()`) in printipi. Why?

I followed  https://github.com/Wallacoloo/printipi/wiki/Adding-support-for-a-new-CPU-architecture
There are "chronoclock.cpp","chronoclock.h","mitpi.cpp","mitpi.h","primitiveiopin.h" and "hardwarescheduler.cpp". under platform/rpi2 directory.
"chronoclock.cpp" and "chronoclock.h" are the same as https://github.com/Wallacoloo/printipi/blob/master/src/platforms/rpi/chronoclock.cpp
https://github.com/Wallacoloo/printipi/blob/master/src/platforms/rpi/chronoclock.h
I changed "primitiveiopin.h" a little.

mitpi.cpp
```
#include "mitpi.h"
#include <sys/mman.h> //for mmap
#include <sys/time.h> 
#include <time.h> //for nanosleep / usleep (if have -std=gnu99)
#include <unistd.h> //for usleep
#include <stdlib.h> //for exit
#include <cassert> //for assert
#include <fcntl.h> //for file opening
#include "common/logging.h"
#include "platforms/auto/primitiveiopin.h"
#include <pigpiod_if.h>
#include <sys/time.h>
#include "platforms/auto/chronoclock.h" //for EventClockT
namespace mitpi {
#define PI_INPUT 0
#define PI_OUTPUT 1
#define PI_LOW 0
#define	PI_HIGH 1
	static volatile uint32_t *gpioBaseMem = nullptr;
	static volatile uint32_t *timerBaseMem = nullptr;
	static volatile uint32_t *gpioPadBaseMem = nullptr;

	static void assertValidPin(PinIntType pin) {
		(void)pin; //unused when assertions are disabled.
		//printf("%d\n",(int)pin);
		assert(pin >= 0 && pin < 64);
	}

	bool init() {
		// Connect to pigpiod
		int status = pigpio_start(0, 0);
		if (status < 0){ LOG("Failed \pigpio_start\n");}

		return 0; //init OK
	}

	void makeOutput(PinIntType pin) {
		assertValidPin(pin);
		set_mode(pin, PI_OUTPUT);
	}
	void makeInput(PinIntType pin) {
		assertValidPin(pin);
		//set_mode(pin, PI_INPUT);
	}
	void setPinHigh(PinIntType pin) {
		assertValidPin(pin);
		gpio_write(pin, PI_HIGH); // 1 is HIGH
	}
[... 299 lines cut ...]
			inline void makeDigitalInput() {
				mitpi::makeInput(pinIdx);
			}
			//read the pin's input value (must call makeDigitalInput beforehand)
			inline IoLevel digitalRead() const {
				return mitpi::readPinState(pinIdx);
			}
			//write a digital value to the pin. IoHigh = 3.3v, IoLow = 0v. Must call makeDigitalOutput beforehand
			inline void digitalWrite(IoLevel lev) {
				mitpi::setPinState(pinIdx, lev);
			}
			//set pwm duty cycle & period (if applicable). Must call makePwmOutput beforehand.
			inline void pwmWrite(float duty, EventClockT::duration desiredPeriod) {
				float minPeriod = std::chrono::duration_cast<std::chrono::duration<float>> (desiredPeriod).count();
				int intMinPeriod = (int)minPeriod;
				uint32_t intDuty = 1000000*duty;
				int flag = hardware_PWM(this->id(), intMinPeriod, intDuty);
				printf("%d",flag);
			}
		};

	}
}
#endif
```

Mokutsuno, 2017-05-03:
[image]